kitspace/kicad_footprints #2: init script fails since `xargs -i` not supported on OSX

brainstorm, 2017-01-09:
I was following the installation instructions and hit this error:

```
(...)
Submodule 'vontrapp/vontrapp.pretty' (https://github.com/vontrapp/vontrapp.pretty.git) registered for path 'vontrapp/vontrapp.pretty'
Submodule 'wiebus/SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty' (https://github.com/wiebus/SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty.git) registered for path 'wiebus/SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty'
Submodule 'xesscorp/RPi_Hat.pretty' (https://github.com/xesscorp/RPi_Hat.pretty.git) registered for path 'xesscorp/RPi_Hat.pretty'
Submodule 'xesscorp/xess.pretty' (https://github.com/xesscorp/xess.pretty) registered for path 'xesscorp/xess.pretty'
xargs: illegal option -- i
xargs: illegal option -- i
usage: xargs [-0opt] [-E eofstr] [-I replstr [-R replacements]] [-J replstr]
             [-L number] [-n number [-x]] [-P maxprocs] [-s size]
             [utility [argument ...]]
```

Apparently `gxargs` the GNU version of `xargs` is no longer in Homebrew, so I switched to GNU parallel and it went fine.

I hesitated to file a pullrequest since I'm not sure if you want to require users to install GNU parallel.

In any case, nice collection of kicad libs, thanks a ton for this effort!

kasbah, 2017-01-09:
Ah, looks like `-i` is deprecated in favor of `-I` in GNU as well. I switched it out with `-I{}` on master just now. Could you test it out on OSX please?

brainstorm, 2017-01-09:
I was actually looking at the Linux manpage and trying `-I{}` instead, but not much success so far:

```
$ git diff init
diff --git a/init b/init
index 47d1af2..b7bfd03 100755
--- a/init
+++ b/init
@@ -7,7 +7,7 @@ max_parallel_tasks=10
 git submodule init

 git submodule | awk '{ print $2 }' \
-  | xargs -i --max-procs=$max_parallel_tasks bash -c \
+  | xargs -I{} --max-procs=$max_parallel_tasks sh -c \
   "echo 'Cloning' {} \
   && git submodule update --quiet --depth 1 {} \
   || git submodule update --quiet {} \

$ ./init
xargs: illegal option -- -
usage: xargs [-0opt] [-E eofstr] [-I replstr [-R replacements]] [-J replstr]
             [-L number] [-n number [-x]] [-P maxprocs] [-s size]
             [utility [argument ...]]
```

And with `-I` as-is, without the `{}`, I'm getting a few of those errors:

```
Cloning {}
error: pathspec '{}' did not match any file(s) known to git.
error: pathspec '{}' did not match any file(s) known to git.
** FAILED ** {}
```

kasbah, 2017-01-09:
Maybe we need to quote the string, or put a space in before it? Or try something other than `{}`?

brainstorm, 2017-01-09:
No dice so far over here... how does it behave on Linux with `-I{}`?

kasbah, 2017-01-09:
It works fine for me.  :confused:. What happens with this?

```
echo "hey" | xargs -I{} sh -c "echo 'you said' '{}'
```

brainstorm, 2017-01-09:
Found it! It was the `--max-processes` which is not supported in OSX either (`-P` does and it's compatible with GNU), this seems to work:

```
$ git diff init
diff --git a/init b/init
index 47d1af2..89a0f96 100755
--- a/init
+++ b/init
@@ -7,7 +7,7 @@ max_parallel_tasks=10
 git submodule init

 git submodule | awk '{ print $2 }' \
-  | xargs -i --max-procs=$max_parallel_tasks bash -c \
+  | xargs -P $max_parallel_tasks -I{} sh -c \
   "echo 'Cloning' {} \
   && git submodule update --quiet --depth 1 {} \
   || git submodule update --quiet {} \
```

brainstorm, 2017-01-09:
Yet there were a few repos failing:

```
$ ./init | grep -i failed
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: no such remote ref b96b1a12917c42b001567578c400fa0f4ce4b88c
Fetched in submodule path 'KiCad/Housings_DIP.pretty', but it did not contain b96b1a12917c42b001567578c400fa0f4ce4b88c. Direct fetching of that commit failed.
error: no such remote ref 6bfa9d84f2c220750c57189f3ffde30f7a7c9a67
Fetched in submodule path 'KiCad/RF_Modules.pretty', but it did not contain 6bfa9d84f2c220750c57189f3ffde30f7a7c9a67. Direct fetching of that commit failed.
error: no such remote ref 6bfa9d84f2c220750c57189f3ffde30f7a7c9a67
Fetched in submodule path 'KiCad/RF_Modules.pretty', but it did not contain 6bfa9d84f2c220750c57189f3ffde30f7a7c9a67. Direct fetching of that commit failed.
** FAILED ** KiCad/RF_Modules.pretty
error: no such remote ref b96b1a12917c42b001567578c400fa0f4ce4b88c
Fetched in submodule path 'KiCad/Housings_DIP.pretty', but it did not contain b96b1a12917c42b001567578c400fa0f4ce4b88c. Direct fetching of that commit failed.
** FAILED ** KiCad/Housings_DIP.pretty
error: pathspec '{}' did not match any file(s) known to git.
error: pathspec '{}' did not match any file(s) known to git.
** FAILED ** {}
```

kasbah, 2017-01-09:
I pushed https://github.com/monostable/kicad_footprints/commit/d99890a6d41aed4a041618383c67d2797ffd6dd3. Could you try with a fresh clone?

brainstorm, 2017-01-09:
Done that right now, all works except towards the end:

```
(...)
Cloning wiebus/SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty
Cloning xesscorp/RPi_Hat.pretty
error: pathspec '{}' did not match any file(s) known to git.
error: pathspec '{}' did not match any file(s) known to git.
** FAILED ** {}
Cloning xesscorp/xess.pretty
$
```

But it looks like it fetched everything right:

```
$ du -hs .
1.4G	.
```

kasbah, 2017-01-09:
That's weird, maybe it's the final newline?

kasbah, 2017-01-09:
What happens with this?

``` 
echo -e 'a\nb\n' | xargs -I{} sh -c 'echo you said {}'
```

brainstorm, 2017-01-09:
All good it seems:

```
$ echo -e 'a\nb\n' | xargs -I{} sh -c 'echo you said {}'
you said a
you said b
```

kasbah, 2017-01-09:
And with `-P`? Maybe I should get hold of a copy of BSD xargs and test instead of this remote debugging.  :laughing:

brainstorm, 2017-01-09:
Hm, but newlines in OSX are funny anyhow:

```
$ echo -e 'a\nb\n\r' | xargs -I{} sh -c 'echo you said {}'
you said a
you said b
you said
$ echo -e 'a\nb\r\n' | xargs -I{} sh -c 'echo you said {}'
you said a
you said b
```

brainstorm, 2017-01-09:
Haha, no worries, this happens with `-P10`:

```
error: pathspec '{}' did not match any file(s) known to git.
error: pathspec '{}' did not match any file(s) known to git.
** FAILED ** {}
$
```

The clone ordering is different now so It's just one of the multiple shells that is getting an empty input, I guess?...

Hmm, what if it's Mac's own `awk`?... Nope, just used `gawk`, the GNU awk version also shows the same behavior...

kasbah, 2017-01-09:
What about?

```
echo -e 'a\r\nb\r\nc\r\nd\r\ne\r\nf\r\ng\r\n' | xargs -I{} -P3 sh -c 'echo you said {}'
```

brainstorm, 2017-01-09:
```
$ echo -e 'a\r\nb\r\nc\r\nd\r\ne\r\nf\r\ng\r\n' | xargs -I{} -P3 sh -c 'echo you said {}'
you said b
you said a
you said c
you said e
you said d
you said f
you said g
```

kasbah, 2017-01-09:
```
echo -e '' | xargs -I{} sh -c 'echo {}'
```

brainstorm, 2017-01-09:
```
$ echo -e '' | xargs -I{} sh -c 'echo {}'
$
```

kasbah, 2017-01-09:
Yeah, makes sense. No idea how it could not replace a `{}` in the actual script.  Did you see my edit above? 

```
echo -e 'a\r\nb\r\nc\r\nd\r\ne\r\nf\r\ng\r\n' | awk '{ print $1 }' | xargs -I{} -P3 sh -c 'echo you said {}'
```

kasbah, 2017-01-09:
The only thing I can think of is outputting to a file and then searching for the culprit.

```bash
$ git submodule | awk '{ print $2 }' > paths
$ wc -l paths
350 paths
$ tail -n 175 paths | xargs -I{} -P10 sh -c   "echo 'Cloning' {} \
  && git submodule update --quiet --depth 1 {} \
  || git submodule update --quiet {} \
  || echo '** FAILED **' {}"
$ head -n 175 paths | xargs -I{} -P10 sh -c   "echo 'Cloning' {} \
  && git submodule update --quiet --depth 1 {} \
  || git submodule update --quiet {} \
  || echo '** FAILED **' {}"
...
```
Does it still happen even when you have already run `./init`?

kasbah, 2017-01-10:
Actually, if you could check that `./update` works when you get a chance, that might be the quicker way to reveal the cause.

brainstorm, 2017-01-10:
The update script resulted in this:

```
Checking wiebus/SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty
sh: line 0: cd: {}: No such file or directory
** FAILED ** {}
```

And the `paths` test went like this:

```
$ git submodule | awk '{ print $2 }' > paths
$ wc -l paths
     350 paths
$ tail -n 175 paths | xargs -I{} -P10 sh -c   "echo 'Cloning' {} \
>   && git submodule update --quiet --depth 1 {} \
>   || git submodule update --quiet {} \
>   || echo '** FAILED **' {}"
(...)
Cloning xesscorp/xess.pretty
error: pathspec '{}' did not match any file(s) known to git.
error: pathspec '{}' did not match any file(s) known to git.
** FAILED ** {}

$ head -n 175 paths | xargs -I{} -P10 sh -c   "echo 'Cloning' {} \
>   && git submodule update --quiet --depth 1 {} \
>   || git submodule update --quiet {} \
>   || echo '** FAILED **' {}"
[All Cloning messages successful]
```

kasbah, 2017-01-10:
So it's in the first half of `paths`! you could now keep cutting it in half until you find it..

`./update`  seems to be the better way to find out though. If you amend the script to use `max_parallel_tasks=1` we might see which one is causing the problem.

brainstorm, 2017-01-10:
Found it! It's just an empty directory issue:

```
Checking vontrapp/vontrapp.pretty
Checking wiebus/SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty
sh: line 0: cd: {}: No such file or directory
** FAILED ** {}
Checking xesscorp/RPi_Hat.pretty
Checking xesscorp/xess.pretty
```

There's nothing in that dir/repo:

```
$ pwd
/Users/romanvg/dev/kicad_modules/kicad_footprints/wiebus/SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty
$ ls
$ ls -a
.  ..
```

kasbah, 2017-01-10:
No, it's empty because it failed! What happens with this?

```
echo wiebus/SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty | xargs -I{} sh -c 'echo you said {}'
```

brainstorm, 2017-01-10:
I did this:

```
$ ls
Romans-MacBook-Air:SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty romanvg$ echo wiebus/SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty | xargs -I{} sh -c 'echo you said {}'
you said wiebus/SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty
Romans-MacBook-Air:SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty romanvg$ git submodule update
Cloning into '/Users/romanvg/dev/kicad_modules/kicad_footprints/wiebus/SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty'...
Submodule path './': checked out '9ed1ee1996fcca713f0dd8fad6a6463a08ae2452'
Romans-MacBook-Air:SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty romanvg$ ls
LICENSE                                                          SOT143_Transistor_Handsoldering_26Jul2012.kicad_mod
README.md                                                        SOT23_Diode_Handsoldering_RevB_14Mar2014.kicad_mod
SC70-6_26Jul2012.kicad_mod                                       SOT23_Diode_RevB_14Mar2014.kicad_mod
SC70-6_Handsoldering_26Jul2012.kicad_mod                         SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty
SOT143R_Diode_Reverse_26Jul2012.kicad_mod                        SOT23_Transistor_BC846_Handsoldering_RevB_14Mar2014.kicad_mod
SOT143R_Diode_Reverse_Handsoldering_26Jul2012.kicad_mod          SOT23_Transistor_BC846_RevB_14Mar2014.kicad_mod
SOT143R_Transistor_Reverse_26Jul2012.kicad_mod                   SOT23_Transistor_Handsoldering_RevB_14Mar2014.kicad_mod
SOT143R_Transistor_Reverse_Handsoldering_26Jul2012.kicad_mod     SOT23_Transistor_RevB_14Mar2014.kicad_mod
SOT143_Diode_26Jul2012.kicad_mod                                 TSOT6-MK06A-Housing_26Jul2012.kicad_mod
SOT143_Diode_Handsoldering_26Jul2012.kicad_mod                   TSOT6-MK06A-Housing_Handsoldering_26Jul2012.kicad_mod
SOT143_Transistor_26Jul2012.kicad_mod
```

But then, still fails and the directory is not empty anymore as per my `git submodule update`:

```
$ ./update
(...)
Checking vontrapp/vontrapp.pretty
Checking wiebus/SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty
sh: line 0: cd: {}: No such file or directory
** FAILED ** {}
```

kasbah, 2017-01-10:
Eeek, it's created a lot of wrong directories/submodules. I would recommend another fresh clone first of all. Still am curious about the commands above.

brainstorm, 2017-01-10:
```
$ echo -e 'wiebus/SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty\r\nb\r\n' | xargs -I{} -P2 sh -c 'echo you said {}'
you said wiebus/SOT23_SOT143_SOT143R_TSOT6_MK06A_SC70-6_Housing_14Mar2014.pretty
you said b
```

kasbah, 2017-01-10:
Well, beats me then. Seems like it should work. Could you paste the `paths` file you created before somewhere?

brainstorm, 2017-01-10:
Sure: http://pastebin.com/6GHtaMrb

kasbah, 2017-01-10:
Thanks, looks fine. I really don't know at this point why this repo is causing the substitution to fail

kasbah, 2017-01-10:
Last ditch effort: I pushed some improved quoting, you could pull that in and give `./update` a try again.

kasbah, 2017-01-10:
One thing I just noticed is that, at 72 characters, the path that is failing is the longest submodule path in the whole repo. Might just be a coincidence though.

brainstorm, 2017-01-13:
Sorry, no clue either. I've been told to switch to parallel which handles these types of issues more gracefully than xargs (unicode, special chars, etc...), but I'm puzzled at it as well.

kasbah, 2017-01-14:
Is `parallel` easily installable on OSX without homebrew and such? Could you share your adapted script with parallel?

If you get a chance could you try the `test-wiebus` branch where I moved the repo to a shorter folder?

brainstorm, 2017-01-20:
No, GNU parallel is not that straightforward to install without `brew`, unfortunately :/

The `test-wiebus` branch fails way more than before:

```
Cloning vontrapp/vontrapp.pretty
Cloning wiebus/Housings_14Mar2014.pretty
error: pathspec '{}' did not match any file(s) known to git.
error: pathspec '{}' did not match any file(s) known to git.
** FAILED ** {}
falling back to deep clone for {}
Cloning xesscorp/RPi_Hat.pretty
error: pathspec '{}' did not match any file(s) known to git.
falling back to deep clone for {}
error: pathspec '{}' did not match any file(s) known to git.
** FAILED ** {}
Cloning xesscorp/xess.pretty
error: pathspec '{}' did not match any file(s) known to git.
falling back to deep clone for {}
error: pathspec '{}' did not match any file(s) known to git.
** FAILED ** {}
error: pathspec '{}' did not match any file(s) known to git.
** FAILED ** {}
```

And does not seem to download all targets since the old installation was 1.4GiB and now it's 1.1GiB:

```
$ du -hs .
1.1G	.
```

kasbah, 2017-01-20:
Whaaaa? I updated the test-wiebus branch, could you try again please.

brainstorm, 2017-01-20:
Wiped and retried again, really bad clones there still... same for `master` actually :-/

```
(...)
error: pathspec '{}' did not match any file(s) known to git.
falling back to deep clone for {}
error: pathspec '{}' did not match any file(s) known to git.
** FAILED ** {}
Cloning xesscorp/xess.pretty
error: pathspec '{}' did not match any file(s) known to git.
falling back to deep clone for {}
error: pathspec '{}' did not match any file(s) known to git.
** FAILED ** {}
error: pathspec '{}' did not match any file(s) known to git.
error: pathspec '{}' did not match any file(s) known to git.
** FAILED ** {}
** FAILED ** {}
(...)
```

kasbah, 2017-01-20:
WTF is going on?

brainstorm, 2017-01-20:
This commit against `init` made things much worse, it seems: https://github.com/monostable/kicad_footprints/commit/23d20aee6b46bdc6110599fba41965df8790dc14

brainstorm, 2017-01-20:
... time to factor in OSX builds in `.travis.yml` I guess :-S : https://docs.travis-ci.com/user/osx-ci-environment/

kasbah, 2017-01-21:
Didn't even think of that! [Trying it now](https://travis-ci.org/monostable/kicad_footprints/builds/193920886).

kasbah, 2017-01-21:
OK! We should be good now. I managed to spin up a FreeBSD Virtualbox and debug the issue with that. I am still not 100% sure what the problem was but splitting out separate scripts solves it and is cleaner anyway. 

I updated master and stable branches, give it a test and close this issue if it solves it for you. I've also included OSX in the Travis builds so that should prevent me from breaking things for you in the future.

brainstorm, 2017-01-22:
Sweet! Now most of the errors are gone, yet there are a few (spurious?) ones while cloning the repos anew... **EDIT:** Two consecutive fresh clones and the 3 errors below persist:

```
Cloning KiCad/Housings_DIP.pretty
Cloning KiCad/Housings_QFP.pretty
error: no such remote ref 80d520028613321675ecb69a6a40cb45ffe19208
```

```
Cloning KiCad/Measurement_Scales.pretty
Cloning KiCad/Mechanical_Sockets.pretty
error: no such remote ref 80d520028613321675ecb69a6a40cb45ffe19208
Fetched in submodule path 'KiCad/Converters_DCDC_ACDC.pretty', but it did not contain 80d520028613321675ecb69a6a40cb45ffe19208. Direct fetching of that commit failed.
 ** Failed ** KiCad/Converters_DCDC_ACDC.pretty
Cloning KiCad/Microwave.pretty
```

```
Cloning xesscorp/RPi_Hat.pretty
Cloning xesscorp/xess.pretty
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: no such remote ref 9f166ffe86d1f3e41b1b8ce0c034773ac0f7983b
Fetched in submodule path 'tylercrumpton/CrumpPrints.pretty', but it did not contain 9f166ffe86d1f3e41b1b8ce0c034773ac0f7983b. Direct fetching of that commit failed.
falling back to deep clone for tylercrumpton/CrumpPrints.pretty
error: no such remote ref 9f166ffe86d1f3e41b1b8ce0c034773ac0f7983b
Fetched in submodule path 'tylercrumpton/CrumpPrints.pretty', but it did not contain 9f166ffe86d1f3e41b1b8ce0c034773ac0f7983b. Direct fetching of that commit failed.
 ** Failed ** tylercrumpton/CrumpPrints.pretty
```

kasbah, 2017-01-22:
Yes, this is a separate issue that arises on all platforms when this repo isn't totally up to date with all the submodules. I thought the "falling back to deep clone" bit would take care of it but it obviously hasn't. I'll open a new issue for it and look into it. Closing this one as the BSD/OSX issues seem to be solved!

closed 2017-01-22 by kasbah