kitspace/kicad_footprints #3: Parsing errors after ./generate-table on OSX

brainstorm, 2017-01-09:
After running `./generate-table` and copying it into the OSX location for it (doing a `.backup` of the former first as advised in the README):

~/Library/Preferences/kicad/fp-lib-table

I'm getting this couple of footprint parsing errors while running CvPcb on KiCad 4.0.5:

```
Errors were encountered loading footprints:
PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' in input/source '/Users/romanvg/dev/kicad_modules/kicad_footprints/lowstar/IPC7351C-Nominal.pretty/CAPC160X80X87L35N.kicad_mod' line 24 offset 14 from /Users/ansible/4.0.5/kicad-mac-packaging-4.0.0/kicad/common/dsnlexer.cpp : Expecting() : line 369
PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' in input/source '/Users/romanvg/dev/kicad_modules/kicad_footprints/adamjvr/KiCAD-OnHand-Lib/OnHand-Components.pretty/BLM15HG601SN1D.kicad_mod' line 16 offset 14 from /Users/ansible/4.0.5/kicad-mac-packaging-4.0.0/kicad/common/dsnlexer.cpp : Expecting() : line 369
```

kasbah, 2017-01-09:
Hmm, seems like this is a problem with the footprints mentioned. The first one opens fine in the FP editor for me though. What happens when you try and open them in the FP editor direct?

brainstorm, 2017-01-09:
No luck with lowstar's one, all the rest load fine 👍 

[image]

kasbah, 2017-01-09:
I looked at the file. It's got a `roundedrect` in it which isn't supported on stable as far as I know. I am not sure why my stable isn't complaining about it.

kasbah, 2017-01-09:
Could add a `./stabilize` command to remove any files that can't be loaded in stable version of KiCAD. [This script](https://github.com/monostable/haskell-kicad-data/blob/master/kicad_footprint_load.py) could work as a start.

brainstorm, 2017-01-10:
Good idea! If kicad can operate in headless mode somehow I would go for TravisCI actually.

kasbah, 2017-01-10:
The linked script is run on TravisCI actually but I still want to include the libraries using new features in general so I was thinking about a command for the user to "stabilize" the their local clone.

Maybe it would be better to keep separate branches though and have CI take care of it. Master could keep adding repos with new features which CI turns into a `stable` branch. People who use stable could then `git clone -b stable`

brainstorm, 2017-01-10:
Hm, I would prefer keeping everything CI'd into master, so that errors are spotted and users are aware of which footprints fail... after all, not all users are using all footprints all the time, so it's not that critical to have a stable branch and it's non-standard/confusing, IMHO, also btw see:

https://github.com/freetronics/freetronics_kicad_library/issues/4

kasbah, 2017-01-10:
Well, if you use a nightly build of KiCAD those footprints will load fine. It makes sense to me if the `master` of this repo tracks the `master` of KiCAD repo and there is a `stable` branch that tracks stable. We can still check both of those in CI with appropriate KiCAD versions.

kasbah, 2017-01-10:
Nice one @brainstorm, I have put up a `stable` branch and both branches are now being checked with Travis. More details are in the README. Thanks for reporting!

closed 2017-01-10 by kasbah