kitspace/kicad_footprints #5: Cloning of out-of date submodules is failing

kasbah, 2017-01-22:
As reported in #2. Not up-to date submodules will fail on init:
```
Cloning xesscorp/RPi_Hat.pretty
Cloning xesscorp/xess.pretty
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: no such remote ref 9f166ffe86d1f3e41b1b8ce0c034773ac0f7983b
Fetched in submodule path 'tylercrumpton/CrumpPrints.pretty', but it did not contain 9f166ffe86d1f3e41b1b8ce0c034773ac0f7983b. Direct fetching of that commit failed.
falling back to deep clone for tylercrumpton/CrumpPrints.pretty
error: no such remote ref 9f166ffe86d1f3e41b1b8ce0c034773ac0f7983b
Fetched in submodule path 'tylercrumpton/CrumpPrints.pretty', but it did not contain 9f166ffe86d1f3e41b1b8ce0c034773ac0f7983b. Direct fetching of that commit failed.
 ** Failed ** tylercrumpton/CrumpPrints.pretty
```

closed 2017-01-22 by kasbah

kasbah, 2017-01-22:
Should be good now, @brainstorm.

brainstorm, 2017-01-22:
Almost!:

```
Cloning KiCad/Housings_DFN_QFN.pretty
error: no such remote ref b5b324a156c1c415b328e9748d02bdc6db7a72c5
Fetched in submodule path 'KiCad/Connectors_Molex.pretty', but it did not contain b5b324a156c1c415b328e9748d02bdc6db7a72c5. Direct fetching of that commit failed.
Submodule out of date: KiCad/Connectors_Molex.pretty
Updating KiCad/Connectors_Molex.pretty
Cloning KiCad/Housings_DIP.pretty
Cloning KiCad/Housings_QFP.pretty
error: no such remote ref 80d520028613321675ecb69a6a40cb45ffe19208
Fetched in submodule path 'KiCad/Converters_DCDC_ACDC.pretty', but it did not contain 80d520028613321675ecb69a6a40cb45ffe19208. Direct fetching of that commit failed.
Submodule out of date: KiCad/Converters_DCDC_ACDC.pretty
Updating KiCad/Converters_DCDC_ACDC.pretty
```

```
Cloning xesscorp/RPi_Hat.pretty
Cloning xesscorp/xess.pretty
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: no such remote ref 9f166ffe86d1f3e41b1b8ce0c034773ac0f7983b
Fetched in submodule path 'tylercrumpton/CrumpPrints.pretty', but it did not contain 9f166ffe86d1f3e41b1b8ce0c034773ac0f7983b. Direct fetching of that commit failed.
Submodule out of date: tylercrumpton/CrumpPrints.pretty
Updating tylercrumpton/CrumpPrints.pretty
```

kasbah, 2017-01-22:
No, yeah, that's fine. It's going to error and fallback to updating.

brainstorm, 2017-01-22:
True that, all good then, thanks a ton @kasbah! :D

kasbah, 2017-01-22:
Maybe I should add in some reassurance so it appears like:
```
Cloning xesscorp/xess.pretty
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: no such remote ref 9f166ffe86d1f3e41b1b8ce0c034773ac0f7983b
Fetched in submodule path 'tylercrumpton/CrumpPrints.pretty', but it did not contain 9f166ffe86d1f3e41b1b8ce0c034773ac0f7983b. Direct fetching of that commit failed.
Submodule 'tylercrumpton/CrumpPrints.pretty' is out of date. But that's OK we'll just update it...
Updating tylercrumpton/CrumpPrints.pretty
```

commit 45e6be0a (2017-01-22): Fall back to updating when initial clone fails

Fixes #5