kitspace/kicad_footprints #6: CvPcb load error on lowstar and adamjvr

brainstorm, 2017-01-26:
Back to Ubuntu from OSX and giving your footprints compilation a good try :)

I just found this:

```
Errors were encountered loading footprints:
PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' in input/source '/home/romanvg/dev/kicad_footprints/adamjvr/KiCAD-OnHand-Lib/OnHand-Components.pretty/C1206.kicad_mod' line 20 offset 14 from /build/kicad-lOwy61/kicad-4.0.4+dfsg1/common/dsnlexer.cpp : Expecting() : line 369
PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' in input/source '/home/romanvg/dev/kicad_footprints/lowstar/IPC7351C-Nominal.pretty/LEDC160X80X65L30N.kicad_mod' line 24 offset 14 from /build/kicad-lOwy61/kicad-4.0.4+dfsg1/common/dsnlexer.cpp : Expecting() : line 369
```

Would it be possible to run the kicad lexer standalone or sth on travis so that those don't ever show up later on?

#idea

kasbah, 2017-01-27:
Those are the two repos [I remove to make the stable branch](https://github.com/monostable/kicad_footprints/commit/c6b9ad70c4d2a0cc1e800312797e52feca3d72ef). You probably want to switch to that. 

I do [load everything into KiCAD with Travis](https://github.com/monostable/kicad_footprints/blob/master/load_all.py) now by the way so if any new parse errors arise it should catch them.

brainstorm, 2017-01-27:
Oh, thanks, sorry, should have realized that earlier...

Just a side note regarding the handling of branches: the most common practice across repos in github is having `master` as stable, by default and other branches as feature/experimental/development, i.e:

http://nvie.com/posts/a-successful-git-branching-model/

I think that would make sense and could save headaches for new adopters of this fantastic repo ;)

closed 2017-01-27 by brainstorm