kitspace/kicad_footprints #7: Adding on Windows?

CryptonZylog, 2017-06-18:
Hi, 

I am running KiCad on Windows. I've successfully generated a `fp-lib-table` file using Python with `generate_table` script and then moved it to `C:\Users\%username%\.config\kicad` directory. However, I am not seeing the footprints in CvPcb when I am loading it. Am I missing something or this is not supported?

For what it's worth, there are several `fp-lib-table` files in KiCad installation directory, however I am not sure what is the right one to replace if that's what is necessary.

Thanks!

kasbah, 2017-06-21:
Hey, you probably don't have the right location for it there but am not exactly sure what it is under Windows. A quick search suggests something like `/Users/[user]/%AppData%/Roaming/kicad`  but you might just have to search for `%appdata%` according to: https://forum.kicad.info/t/kicad-github-pcb-footprint-libraries/119/7

Remember you have to restart KiCad for it to take effect.

CryptonZylog, 2017-06-21:
Looks like the correct path, that is `C:\Users\%username%\AppData\Roaming\kicad`. There's a fp-lib-table file there already. When I replaced it with the generated one, I am not getting the following error when launching CvPcb.

[image]

kasbah, 2017-06-21:
Can you paste what's in your new fp-lib-table file?

CryptonZylog, 2017-06-21:
[Here it is](https://gist.github.com/CryptonZylog/122a69be853a9741ab02b04b8731bf7e)

kasbah, 2017-06-22:
That looks fine to me, I can't figure out what the issue is. Do you have a backup of your old fp-lib-table?

CryptonZylog, 2017-06-22:
[Here's the default one installed with kicad](https://gist.github.com/CryptonZylog/e4010e2c2da9d5f48d1c53bced90d3cd)

kasbah, 2017-06-22:
And if you put that back everything is fine?

kasbah, 2017-06-27:
Let me know if you have any luck with fixing this, I would love to add instructions for Windows users.

CryptonZylog, 2017-06-27:
Sorry, been out the past couple of days.

Yes, restoring the original file and CvPcb launches just fine.

kasbah, 2017-06-27:
Hmm, maybe the way Python writes the paths on Windows (with `/`) isn't supported by KiCad. Could you add one of the libs on you hard disk through the GUI and paste the updated fp-lib-table?

kasbah, 2017-07-05:
@CryptonZylog, I managed to get some time on a Windows 7 machine and test things out. It did indeed seem to be and issue with paths which KiCad expects to use a Unix style path seperator but Python uses a Windows style. Could you `git fetch origin && git checkout origin/windows` and test with that?

CryptonZylog, 2017-07-05:
Still getting the same error.
The `fp-lib-table` I noticed does have forward slashes now. I wonder if there's an issue with running KiCad on Win 10?

kasbah, 2017-07-05:
I have never heard that but haven't tried it myself. It seems like it works fine if you don't use the script to generate the table right? I guess I have to test on Windows 10 myself though. 

The error message you posted above, which uses Cyrillic alphabet, it's saying it expected a `(` at line 1 character 1 right? Do any footprints show up at all when you get that error? 

Would be able to open the working table in a text editor and copy one of the `(lib ...` expressions across from the not working one and see what that does?

CryptonZylog, 2017-07-05:
I've ran the python script again to re-generate the fp-lib-table file and noticed the forward slashes. Correct, it's not liking the line 1, char 1 for some reason. No footprints show up and CvPcb occasionally crashes.

>Would be able to open the working table in a text editor and copy one of the (lib ... expressions across from the not working one and see what that does?

Let me get back to you soon on that one.

kasbah, 2017-07-05:
Ok, thanks. Maybe it's a unicode or other encoding issue, if you have installed, or could install something (git-bash or Ubuntu Sub-system for Windows), which has a [`file`](http://manpages.ubuntu.com/manpages/xenial/en/man1/file.1.html) program or have some other way of finding out the encoding of the different files? If I run file on my fp-lib-table for instance it says: 

```
$ file fp-lib-table
fp-lib-table: ASCII text
```

If I put a Greek character in there:

```
fp-lib-table: UTF-8 Unicode text
```

CryptonZylog, 2017-07-05:
Have Cygwin installed and this is what I got:

```
$ file fp-lib-table
fp-lib-table: Little-endian UTF-16 Unicode text, with CRLF, CR line terminators
```

CryptonZylog, 2017-07-05:
Just made some progress on this! Good thought about encoding and line endings.

I opened fp-lib-table in Notepad++ then decided to re-encode with UTF-8 without BOM **and** change line endings to unix-style. CvPcb after a long wait, finally loaded. And I could even load some footprints!

However, it seems few errors now still creep up. Got an error window with the following:

```
PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' в вводе/источнике 'C:\Users\crypton\Documents\Code\kicad_footprints\lowstar\IPC7351C-Nominal.pretty\CAPC160X80X87L35N.kicad_mod' строка 24 смещение 14 из C:/Jenkins/workspace/windows-kicad-msys2-stable/src/kicad/common/dsnlexer.cpp : Expecting() : line 369
PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' в вводе/источнике 'C:\Users\crypton\Documents\Code\kicad_footprints\PatrickBaus\footprints.pretty\jumper_solder_large.kicad_mod' строка 9 смещение 14 из C:/Jenkins/workspace/windows-kicad-msys2-stable/src/kicad/common/dsnlexer.cpp : Expecting() : line 369
PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' в вводе/источнике 'C:\Users\crypton\Documents\Code\kicad_footprints\Toroid-io\toroid-kicad.pretty\U-PDFN-8.kicad_mod' строка 13 смещение 14 из C:/Jenkins/workspace/windows-kicad-msys2-stable/src/kicad/common/dsnlexer.cpp : Expecting() : line 369
PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' в вводе/источнике 'C:\Users\crypton\Documents\Code\kicad_footprints\adamjvr\KiCAD-OnHand-Lib\OnHand-Components.pretty\BLM15HG601SN1D.kicad_mod' строка 16 смещение 14 из C:/Jenkins/workspace/windows-kicad-msys2-stable/src/kicad/common/dsnlexer.cpp : Expecting() : line 369
```

kasbah, 2017-07-05:
Nice work! What shell were you using to run the generate_table script? When
I was testing on Win7 I was using git-bash shell, so maybe that's why I
didn't get the error then.

I think those last errors might be because you are running stable KiCad but
are still including libraries with unstable features (rounded rectangle). I
think if you switch to the `windows` branch and re-generate it should fix
it.

CryptonZylog, 2017-07-05:
Was using PowerShell, then ``python generate_table > fp-lib-table``

kasbah, 2017-07-05:
I pushed a version to the `windows` branch that uses `codec.open` instead of a shell redirect. Could you try that please? Could you give me the PowerShell version of [these instructions](https://github.com/monostable/kicad_footprints/blob/090d6e40060070221b32834c18556f3bd99dbd31/README.md#windows-using-git-bash)?

kasbah, 2017-10-07:
Hey @CryptonZylog I expect the issue is resolved with #9. Would be good to get confirmation though.

closed 2017-10-07 by kasbah