kitspace/kicad_footprints #13: Path is /cygdrive/c/Users instead of c:/Users

stanleyseow, 2018-04-14:
I'm using the window branch, when generating fp_lib_table using Git Bash, it becomes :- 

  (lib (name "digikey/digikey-footprints.pretty")(type KiCad) (uri "/cygdrive/c/Users/stanley/Documents/Github/kicad_footprints/digikey/digikey-footprints.pretty") (options "") (descr ""))

This is a Win 7 32-bit version..

I have another  Win 7 64-bit version with no such issues ...
Any idea why ??

kasbah, 2018-04-14:
I am not too familiar with Windows and especially Bash for windows. Do you maybe also have Cygwin installed on the first machine?

stanleyseow, 2018-04-14:
You are right, I have a cygwin installed on the first machine. Once I uninstalled it, the issue goes away. Closing the issue now.. 

Thanks!!!!

closed 2018-04-14 by stanleyseow